codebndr/Ariadne-Bootloader #20: Ariadne installation via Arduino Boards Manager

per1234, 2015-05-08:
I have created a working implementation of installation of Ariadne to the Arduino IDE via Boards Manager. The URL is:
https://raw.githubusercontent.com/per1234/Ariadne-Bootloader/boards-manager-installation-test/package_codebendercc_ariadne-bootloader_index.json
This can be added to Arduino IDE 1.6.4 in File > Preferences > Additional Boards Manager URLs. Then Ariadne Bootloader will appear as one of the options for installation under Tools > Board > Boards Manager...

I would like to create a pull request to add this capability to Ariadne but first I need the installation file to be available. The most common way of doing this seems to be a Github release. My installation file is here:
https://github.com/per1234/Ariadne-Bootloader/releases/download/v1.0.0/ariadne.zip
You can download mine to use or create your own. If you create your own notice that the folder structure has to be changed slightly.

For examples of how others have set this up see:
https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls

Thanks, Per

LoathingKernel, 2015-05-08:
thank you for doing this. Right now i do not have access to my computer nor i  have stable access to the internet (i am writing this on a nokia c-5) and i will not do so for at least 2 months. I think you should use your own repo for now until i am able to act accordingly.

per1234, 2015-05-08:
Ok, I will do that. Let me know whenever you want to continue. Until then the URL listed above will be the one people can use for installation via Boards Manager. Thanks for your work on this project! I use it almost daily.

q2dg, 2015-08-26:
Hello. Any news about it? It should be fantastic if it goes forward.
Thanks

q2dg, 2015-10-21:
Please...

per1234, 2015-10-21:
@q2dg I agree it would be great to see Boards Manager install added to this repository but in the meantime you can always use the URL shown in my first comment.

LoathingKernel, 2015-10-26:
I will have time from now on to merge any pull requests. @per1234, If you have some time later this week, send me a quick email explaining your changes to help me navigate through your changes/branches faster. It would be a great deal of help.

per1234, 2015-10-26:
Welcome back! I will put together some information on making this happen and send you an email soon.
Per

LoathingKernel, 2016-05-10:
This issue was moved to LoathingKernel/Ariadne-Bootloader#3

closed 2016-05-10 by LoathingKernel