codebndr/Ariadne-Bootloader #23: Bootloader hanging on field application -2560

andrebstv, 2016-02-20:
Hi,

I couldn`t help but notice the bootloader is not protected by a WDT. I have a device on the field that from time to time hangs, and as I can see from logs writen on a SD card It does in the bootloader section. (it shows a "reboot" string on the log and then no "Power ON" string which is the very first thing it does on app start).

I`ve changed the bootloader code to include proper wdt protection, but I`m unable to compile it... It may sound like a noob error, but it fails on the memcpy_PF... I`ve seen previous ppl having this error but no solution to it.

I do compile on windows, using the WinAVR toolchain.

Where should I change on the makefile so I can compile it? Tried everywhere and still got this error.

Note that I would gladly post the wdt fix once I`ve tested.

LoathingKernel, 2016-02-22:
What I think that will fix your issue is downloading the AVR toolchain from here http://www.atmel.com/tools/ATMELAVRTOOLCHAINFORWINDOWS.aspx

andrebstv, 2016-02-27:
I solved the compiling issue. Apparently you need to compile the target atmega328 to generate some .o files that are used when compiling the atmega2560 target and the debug2560 target.

It`s a bug on the makefile. I won`t try to fix that. I might break something... my makefileism`s skill are not good.

I will do some more testing with the modded bootloader, that includes the wdt, and if successful post the fixes.

andrebstv, 2016-02-27:
Correction: it's actually a difference on the  pgmspace.h from the winavr toolchain vs the atmel toolchain.
The make file is correct.

andrebstv, 2016-03-28:
Created the pull request. Still sometimes the signature seems to be corrupted and the bootloader doesn`t start the application.... a fresh upload solves this. Main thing is that the board is no longer hangable.

andrebstv, 2016-04-01:
The problem was fixed, but now everytime the application resets it hangs on the bootloader. Never allowing the user app to run. I even tried to enforce the eeprom signature write on the user app, to trick the bootloader but no avail.

A fresh upload, that is a firmware "update", fixes the problem...Probably deactivating the eeprom signature check fixes it... but that seems a bit like a cludge.

Help please?

LoathingKernel, 2016-04-02:
Can you give me more information? Have you tried increasing the WDT timer to give more time for the bootloader to load?

LoathingKernel, 2016-05-10:
This issue was moved to LoathingKernel/Ariadne-Bootloader#5

closed 2016-05-10 by LoathingKernel