codebndr/Ariadne-Bootloader #25: TFTP upload fails with the message  "Timeout occurred,     Connect request failed"

intelligentandgoodlooking, 2016-03-11:
I have burned Ariadne Aootloader to an arduino mega2560 using another arduino mega 2560 as an ISP, using [Nick Gammon's Atmega_Board_Programmer] (https://github.com/nickgammon/arduino_sketches). I followed Nick's instructions, to  [convert the .hex bootloader files on disk for use with his sketch](http://www.gammon.com.au/forum/?id=11635). Everything reports a successful burn, and I can upload sketches via serial no problem (blink, fade etc). 

However successful this sounds, I cannot upload via TFTP :
- I am connected to the Ethernet shield directly via Ethernet cable (no router)
- I have uploaded the WriteNetworkSettings sketch and changed the gateway IP to the same IP as my windows PC.
- I can ping the device on the default IP address 192.168.1.128
- I have converted the fade.hex to bin using avr-objcopy.exe
-  I am resetting the board and waiting for about 10sec (for ethernet to settle) and then typing into a command prompt:   tftp -i 192.168.1.128 PUT Fade.bin
- I have led pin 13 flashing at about 5Hz for about 20 sec and then it goes into the sketch which i uploaded via serial - currently I'm using the standard fade example.

TFTP still fails. What have I missed?
suddenly my username doesnt seem so clever :(

intelligentandgoodlooking, 2016-04-13:
I have managed to get the Arduino as ISP working with another mega, and I have successfully burned the arduino2560 (with debug)  using the Burn Bootloader function in the Arduino IDE. 
Serial flashing works but I still have no success when flashing via TFTP. I have tried in command line (windows 7)and in Ubuntu using atftp. Both methods timeout.

I have pasted the bootloader output at the bottom. You can see the TFTP server is receiving the same packet 4 times. It seems like my tftp client is not receiving the ACK.

I read in the readme about using "ip_conntrack_tftp" that will allow the client to receive ACK, but does this only apply to Redhat distros? 
Is there anything special required for DOS?

Main: Ariadne for Arduino Mega2560, Version 0.5
 Net: EEPROM settings
 Net: 	Address: 0xC0.0xA8.0x01.0x80
 Net: 	 Subnet: 0xFF.0xFF.0xFF.0x00
 Net: 	Gateway: 0xC0.0xA8.0x01.0x02
 Net: 	    MAC: 0x90.0xA2.0xDA.0x0F.0xDF.0xB7
 Net: Network init done
Tftp: Init socket to port 0x0045
Tftp: TFTP server init done
Tftp: 	DataPort: 0xB779
Tftp: Processing packet of size 0x001C
Tftp: Set up return address
Tftp: This is block 0x626C with opcode 0x0002 and data length 0x0010
Tftp: Write request
Tftp: Init socket to port 0xB779
Tftp: Changed to port 0xB779
Tftp: Sent ACK
Tftp: Response sent
Tftp: Init socket to port 0x0045
Tftp: TFTP server init done
Tftp: 	DataPort: 0xB779
Tftp: Processing packet of size 0x001C
Tftp: Set up return address
Tftp: This is block 0x626C with opcode 0x0002 and data length 0x0010
Tftp: Write request
Tftp: Init socket to port 0xB779
Tftp: Changed to port 0xB779
Tftp: Sent ACK
Tftp: Response sent

LoathingKernel, 2016-04-13:
What is the IP of the machine you are trying to upload the program from and what is the Gateway set in the network settings of the arduino?

LoathingKernel, 2016-04-13:
Since you are directly connected to the arduino, those two should match.

intelligentandgoodlooking, 2016-04-13:
I had both my PC and the arduino connected to a router with Ip address 192.168.1.2
I have now removed the router and connected directly, changing the gateway address as you suggested. 
I have found that if I get a ping response, the tftp upload will work, but I only get a ping response sometimes.
If I run WriteNetworkSettings with the line NetEEPROM.writeImgBad();commented out, it causes the wdt to reset after 4 seconds. 
The ping only works if I wait for at least cycle of: "start user app" and WDT reset. Sometimes it takes more than 1 reset.

LoathingKernel, 2016-04-13:
Can you post here the network conf of your Arduino and the machine you are trying to upload from?

>  I have now removed the router and connected directly, changing the gateway address as you suggested. 
Flashing through the router should be fine as long as the gateway on the arduino is the router's ip.

> I have found that if I get a ping response, the tftp upload will work, but I only get a ping response sometimes.
This is weird, ping should work out of the box if the network is configured correctly. The W5100 chip replys to pings on its own, the bootloader has nothing to do with it other than configuring the network interface.

> If I run WriteNetworkSettings with the line NetEEPROM.writeImgBad();commented out, it causes the wdt to reset after 4 seconds.
This is expected to happen as the bootloader tries to run the product of an icomplete upload. Is that the case here?

> The ping only works if I wait for at least cycle of: "start user app" and WDT reset. Sometimes it takes more than 1 reset.
Is there a chance that your shield has some kind of issue? I do not want to offload the problem, I am just trying to rule out any other case because this seems random and I do not have any ideas as to what might be causing it at the moment.

intelligentandgoodlooking, 2016-04-13:
I believe I have characterized the problem reliably, and achieved a workable solution.

The problem with not receiving the ACK packets was because my PC had the gateway 192.168.1.1 which it gets via DHCP from my other router downstairs, even though the both the arduino and the PC are physically connected to the router with IP 192.168.1.2. 
Rookie mistake.

The problem with the no ping response seems to only occur when I manually reset (press the reset button). If I use EthernetReset or anything else which causes a WDT reset, I get a ping response every time, and I can upload reliably via TFTP from either Windows or Ubuntu.

I will try to get another set of harware just to test... like you said, could be a shaky ethernet shield.

intelligentandgoodlooking, 2016-04-13:
I have burned the bootloader to an iBoard pro from itead Studios and it works fine using the reset button, and the EthernetReset also works. 

The iBoard Pro is Chinese made board with a mega2560 chip and W5100 chip connected the same way as the ethernet shield and arduino mega combination.

So it turns out i have some bad hardware. Not sure if its the ethernet shield or the mega... or both.
Will have to do some more investigating. Wow... I have lost about 4 full days of time over some bad hardware... lesson learned.

Thanks for your help Stelios

LoathingKernel, 2016-04-13:
Can you compile the bootloader on your own? If so you could try changing this line to give more time to the shield to initialize after a hard reset.

[Relevant line](https://github.com/codebendercc/Ariadne-Bootloader/blob/3bca71f085b41f46556442a06384f49142709e87/hardware/ariadne/bootloaders/ariadne/main.c#L41)

LoathingKernel, 2016-04-13:
Great to hear that it is working for you now. 

Have you, by any chance, flashed the bootloader to the arduino without disconnecting the shield first? I know, you are not supposed to do that, but I have done it quite a few times and one shield started to misbehave after a while.

intelligentandgoodlooking, 2016-04-14:
I have never compiled the bootloader on my own before... I assume its to do with the makefile? I'm still a  bit new to linux. I'd be keen to have a go, but right now I need to finish the project this all contributes to... got a meeting with the client tomorrow!

intelligentandgoodlooking, 2016-04-14:
When I burned the bootloader on the arduino mega and shield I was removing the shield each time so that I had access to the SPI pins. Interestingly that was the one that was giving problems.
When I burned the bootloader to the iBoard Pro, the W5100 was still connected as it is part of the same board. That one works fine.

LoathingKernel, 2016-04-14:
I just had an idea, can you check if the W5100 gets relatively hot in any of the boards while in the bootloader waiting to be programmed?

intelligentandgoodlooking, 2016-04-14:
I cant test it on the mega/shield combo because I have a POE module which covers the W5100.
On the iBoard Pro (in which the bootloader appears to be working fine) the W5100 does feel like it is getting warm, yes.

closed 2016-04-29 by LoathingKernel

arvabhatrajesh, 2019-12-07:
board: mega2560
Eshield: w5500
I was flashed Ariadne-Bootloader using USBASP. 
File: ariadne_atmega2560_w5500.hex
flashing to mega2560 is done
after i changed ip address using Writenetworksettings 
ip address is pinging 
tracert also working
tftp -i 192.168.1.128 PUT atmega2560_blink13.bin
after 20 sec it shows like " CONNECT REQUEST FAILED"

anyone suggest me for better way

Thank you.

loathingKernel, 2019-12-07:
Are you using the Arduino Ethernet Shield with the W5500? You might be affected by this bug https://github.com/loathingKernel/ariadne-bootloader/issues/37. Use a later version of Ariadne from my [personal repository](https://github.com/loathingKernel/ariadne-bootloader) or from an [up-to-date fork](https://github.com/embeddedartistry/ariadne-bootloader)