codebndr/Ariadne-Bootloader #30: Bootloader hangs on Mega2560

Hagie0815, 2016-05-27:
Hi all,

i think this ist not a real issue, and should be solved in short time.

I am normaly using the Mega2560-Board without any Bootloader. but in one project i need the possibility to do a sketch-update via Ethernet.
So the Ariadne-Bootloader was a perfect solution!

It is running perfect!

But, when there is connected a PC to "Serial" it hangs, when both (PC and Arduino) starting at the same time. i think the pc is sending some trash, and the bootloader is thinking "hey cool, i have to receive a sketch"... 
Disconnecting the PC from Serial is solving this problem. but i need this connection for controlling/debugging.

I don't need the serial-update feature. is it possible to disable it, or to get a version of this bootloader without serial functions?

Thanks a lot in advance.

cu
Hagen

LoathingKernel, 2016-05-27:
Yeah, you can disable it, you just have to commend the serialPoll function in main.c. It is a quick and dirty fix but it should do the trick. Right now the makefile doesn't support building without serial but it should be something trivial to add.

per1234, 2016-05-27:
I've been meaning to check if disabling all the serial code would allow Ariadne to use a smaller boot flash section size. Of course this will change the procedure for writing the network settings to EEPROM since you couldn't upload the WriteNetworkSettings NetEEPROM example over Serial once Ariadne has been installed but that shouldn't be a problem, it would only require adding some documentation of how to accomplish it.

Hagie0815, 2016-05-28:
@LoathingKernel 
Thanks! Thats excat what i needed! :+1: 

an offtopic-question: i want to change the LED-Port to Digital-7. Where can i change this?

cu
Hagen

LoathingKernel, 2016-05-28:
@Hagie0815 
You can change that in spi.h. There you will find the pin definitions for all the supported boards and you can change it accordingly.

@per1234 
Simply commenting out serialPoll, will give you a rough idea of what the size of the bootloader without serial would be thanks to the linker options. The --gc-sections as well as -flto strip unneeded binary code sections from the final binary at link time, so anything that is not used is not taking any space, among other things. The only reason I am a bit hesitant to actually add that to the makefile is that right now we have 9 different targets for each board+wiznet compination plus 1 for ethernet plus another 9 for debugging. Adding that, it would be another 9 targets with disabled serial. It is going to be a mess. so any ideas/suggestions/"Stelios, you are stupid, just add it, it is not a problem" is appreciated.

Hagie0815, 2016-05-28:
@LoathingKernel 
Thanks a lot. now it is exactly working, how i need it!

thanks LoathingKernel and per1234!

cu
Hagen

per1234, 2016-05-29:
Here's the output of avr-size comparing the standard targets with ones that have serial uploading disabled, all compiled on avr-gcc 6.1.0:
```
   text	   data	    bss	    dec	    hex	filename
      0	   2982	      0	   2982	    ba6	ariadne_atmega1284_w5100.hex
      0	   2310	      0	   2310	    906	ariadne_atmega1284_w5100_noserial.hex
      0	   3028	      0	   3028	    bd4	ariadne_atmega1284_w5200.hex
      0	   2356	      0	   2356	    934	ariadne_atmega1284_w5200_noserial.hex
      0	   2986	      0	   2986	    baa	ariadne_atmega1284_w5500.hex
      0	   2312	      0	   2312	    908	ariadne_atmega1284_w5500_noserial.hex
      0	   3902	      0	   3902	    f3e	ariadne_atmega2560_w5100.hex
      0	   2412	      0	   2412	    96c	ariadne_atmega2560_w5100_noserial.hex
      0	   3948	      0	   3948	    f6c	ariadne_atmega2560_w5200.hex
      0	   2458	      0	   2458	    99a	ariadne_atmega2560_w5200_noserial.hex
      0	   3906	      0	   3906	    f42	ariadne_atmega2560_w5500.hex
      0	   2414	      0	   2414	    96e	ariadne_atmega2560_w5500_noserial.hex
      0	   2646	      0	   2646	    a56	ariadne_atmega328_ethernet.hex
      0	   2082	      0	   2082	    822	ariadne_atmega328_ethernet_noserial.hex
      0	   2642	      0	   2642	    a52	ariadne_atmega328_w5100.hex
      0	   2072	      0	   2072	    818	ariadne_atmega328_w5100_noserial.hex
      0	   2688	      0	   2688	    a80	ariadne_atmega328_w5200.hex
      0	   2118	      0	   2118	    846	ariadne_atmega328_w5200_noserial.hex
      0	   2646	      0	   2646	    a56	ariadne_atmega328_w5500.hex
      0	   2074	      0	   2074	    81a	ariadne_atmega328_w5500_noserial.hex
      0	   4286	      0	   4286	   10be	ariadne_debug1284_w5100.hex
      0	   3668	      0	   3668	    e54	ariadne_debug1284_w5100_noserial.hex
      0	   4604	      0	   4604	   11fc	ariadne_debug1284_w5200.hex
      0	   3986	      0	   3986	    f92	ariadne_debug1284_w5200_noserial.hex
      0	   4562	      0	   4562	   11d2	ariadne_debug1284_w5500.hex
      0	   3944	      0	   3944	    f68	ariadne_debug1284_w5500_noserial.hex
      0	   5208	      0	   5208	   1458	ariadne_debug2560_w5100.hex
      0	   3770	      0	   3770	    eba	ariadne_debug2560_w5100_noserial.hex
      0	   5256	      0	   5256	   1488	ariadne_debug2560_w5200.hex
      0	   3816	      0	   3816	    ee8	ariadne_debug2560_w5200_noserial.hex
      0	   5214	      0	   5214	   145e	ariadne_debug2560_w5500.hex
      0	   3774	      0	   3774	    ebe	ariadne_debug2560_w5500_noserial.hex
      0	   3944	      0	   3944	    f68	ariadne_debug328_w5100.hex
      0	   3426	      0	   3426	    d62	ariadne_debug328_w5100_noserial.hex
      0	   3990	      0	   3990	    f96	ariadne_debug328_w5200.hex
      0	   3472	      0	   3472	    d90	ariadne_debug328_w5200_noserial.hex
      0	   3948	      0	   3948	    f6c	ariadne_debug328_w5500.hex
      0	   3430	      0	   3430	    d66	ariadne_debug328_w5500_noserial.hex
```
I was able to find a few other lines of code(https://github.com/per1234/Ariadne-Bootloader/commit/0a240769efc7d9d0ce58e06597de1d9fde72beb8) to disable to get the size smaller but there might be more I didn't find. ariadne_atmega328_w5100_noserial.hex and ariadne_atmega328_ethernet_noserial.hex are close to fitting in 2 KB but I'm not sure how difficult it would be to free up that space. I was able to accomplish it by removing the code to write the version to EEPROM.

Freeing up 2 KB of flash on the ATmega328P based boards could be a very big improvement for some people, especially since the type of code running on projects that use Ariadne is likely to be pretty large. Barely squeezing into the section could be limiting to adding new features to Ariadne though.

I definitely do see your point LoathingKernel about adding more options greatly increasing the amount of work required to test and support this project. My fork has even more custom menus, etc. so I've experienced this first hand. I'm willing to take responsibility for some of the testing. I guess there are two different approaches that can be taken:

1. Offering a more limited yet thoroughly tested product
1. Providing many options, some of which may not be fully tested and relying on the users to report any bugs that may slip by.

I think the second option is perfectly valid in this sort of a volunteer-run free open source project as long as there is a reasonable attempt to make sure the most commonly used configurations are functional.

My personal preference is that Ariadne have as many features as possible without causing significant bloat. I never use serial uploads on my devices that run Ariadne so I think this feature would be useful as a custom Tools menu option in the IDE.

LoathingKernel, 2016-05-29:
Quick reply: 

Add the following declaration in main.c
`int main(void) __attribute__ ((OS_main)) __attribute__ ((section (".init9")));`
in the place of the previously commented out `main` declaration and magic will happen.

per1234, 2016-05-29:
post-magic results:
```
   text	   data	    bss	    dec	    hex	filename
      0	   2910	      0	   2910	    b5e	ariadne_atmega1284_w5100.hex
      0	   2238	      0	   2238	    8be	ariadne_atmega1284_w5100_noserial.hex
      0	   2956	      0	   2956	    b8c	ariadne_atmega1284_w5200.hex
      0	   2284	      0	   2284	    8ec	ariadne_atmega1284_w5200_noserial.hex
      0	   2914	      0	   2914	    b62	ariadne_atmega1284_w5500.hex
      0	   2240	      0	   2240	    8c0	ariadne_atmega1284_w5500_noserial.hex
      0	   3830	      0	   3830	    ef6	ariadne_atmega2560_w5100.hex
      0	   2340	      0	   2340	    924	ariadne_atmega2560_w5100_noserial.hex
      0	   3876	      0	   3876	    f24	ariadne_atmega2560_w5200.hex
      0	   2386	      0	   2386	    952	ariadne_atmega2560_w5200_noserial.hex
      0	   3834	      0	   3834	    efa	ariadne_atmega2560_w5500.hex
      0	   2342	      0	   2342	    926	ariadne_atmega2560_w5500_noserial.hex
      0	   2574	      0	   2574	    a0e	ariadne_atmega328_ethernet.hex
      0	   2018	      0	   2018	    7e2	ariadne_atmega328_ethernet_noserial.hex
      0	   2570	      0	   2570	    a0a	ariadne_atmega328_w5100.hex
      0	   2012	      0	   2012	    7dc	ariadne_atmega328_w5100_noserial.hex
      0	   2616	      0	   2616	    a38	ariadne_atmega328_w5200.hex
      0	   2058	      0	   2058	    80a	ariadne_atmega328_w5200_noserial.hex
      0	   2574	      0	   2574	    a0e	ariadne_atmega328_w5500.hex
      0	   2014	      0	   2014	    7de	ariadne_atmega328_w5500_noserial.hex
      0	   4214	      0	   4214	   1076	ariadne_debug1284_w5100.hex
      0	   3596	      0	   3596	    e0c	ariadne_debug1284_w5100_noserial.hex
      0	   4532	      0	   4532	   11b4	ariadne_debug1284_w5200.hex
      0	   3914	      0	   3914	    f4a	ariadne_debug1284_w5200_noserial.hex
      0	   4490	      0	   4490	   118a	ariadne_debug1284_w5500.hex
      0	   3872	      0	   3872	    f20	ariadne_debug1284_w5500_noserial.hex
      0	   5136	      0	   5136	   1410	ariadne_debug2560_w5100.hex
      0	   3698	      0	   3698	    e72	ariadne_debug2560_w5100_noserial.hex
      0	   5190	      0	   5190	   1446	ariadne_debug2560_w5200.hex
      0	   3744	      0	   3744	    ea0	ariadne_debug2560_w5200_noserial.hex
      0	   5140	      0	   5140	   1414	ariadne_debug2560_w5500.hex
      0	   3702	      0	   3702	    e76	ariadne_debug2560_w5500_noserial.hex
      0	   3872	      0	   3872	    f20	ariadne_debug328_w5100.hex
      0	   3366	      0	   3366	    d26	ariadne_debug328_w5100_noserial.hex
      0	   3918	      0	   3918	    f4e	ariadne_debug328_w5200.hex
      0	   3412	      0	   3412	    d54	ariadne_debug328_w5200_noserial.hex
      0	   3876	      0	   3876	    f24	ariadne_debug328_w5500.hex
      0	   3370	      0	   3370	    d2a	ariadne_debug328_w5500_noserial.hex
```

LoathingKernel, 2016-05-29:
So yeah, at least for the atmega328 without the serial and everything else as it is, my version fits in 2KB space, even though just barely, and so does yours.

closed 2016-06-03 by LoathingKernel