codebndr/Ariadne-Bootloader #34: Bootloader burned successfully, Failed to upload bin via TFTP.

whitesn, 2016-08-18:
Hello, so I am able to burn the bootloader and upload the "ReadNetworkSettings" sketch using Serial USB Comm.

I am also able to ping the default IP when the Arduino is powered and connected (192.168.1.128)

Then I tried to upload the sample "blink" ( utilities\tests\blink\atmega328_blink9.bin ) by resetting the Arduino Board (LED13 now blinks quickly), then type "tftp -i 192.168.1.128 PUT atmega328_blink9.bin" (I already cd'd to the folder). After some times, it says "Connect request failed", however, now my Arduino is not able to load the previous sketch and will keep blinking quickly (even after resetting, maybe waiting for incoming upload request?)

The previously uploaded sketch is now gone, but I am still able to upload sketches via serial. What should I do?.. (Windows 10 64bit, Arduino 1.6.7)

FunnyPaul, 2016-09-05:
Hi, I have also had problems with the TFTP tool of Windows. Try it once [http://tftpd32.jounin.net](http://tftpd32.jounin.net)  with the tool. And adjust or turn off the firewall.

whitesn, 2016-09-14:
Thanks @FunnyPaul , I'll get back after trying this out later.

LoathingKernel, 2017-05-10:
This issue was moved to LoathingKernel/Ariadne-Bootloader#18

closed 2017-05-10 by LoathingKernel