codebndr/Ariadne-Bootloader #35: getting error when flashing with tftp

flintec, 2016-10-21:
i get a error from the arduino when im flashing using the tftp client take a look at this.
this is the output fom the serial port.

ˆC¡*¥¹é Ariadne for Arduino Uno, Version 0.5
 Net: EEPROM settings
 Net: 	Address: 0xC0.0xA8.0x01.0x78
 Net: 	 Subnet: 0xFF.0xFF.0xFF.0x00
 Net: 	Gateway: 0xC0.0xA8.0x01.0x01
 Net: 	    MAC: 0x12.0x34.0x56.0x78.0x9A.0xBC
 Net: Network init done
Tftp: Init socket to port 0x0045
Tftp: TFTP server init done
Tftp: 	DataPort: 0xB779
Tftp: Processing packet of size 0x001E
Tftp: Set up return address
Tftp: This is block 0x426C with opcode 0x0002 and data length 0x0012
Tftp: Write request
Tftp: Init socket to port 0xB779
Tftp: Changed to port 0xB779
Tftp: Sent ACK
Tftp: Response sent
Tftp: Init socket to port 0x0045
Tftp: TFTP server init done
Tftp: 	DataPort: 0xB779
Tftp: Processing packet of size 0x001E
Tftp: Set up return address
Tftp: This is block 0x426C with opcode 0x0002 and data length 0x0012
Tftp: Write request
Tftp: Init socket to port 0xB779
Tftp: Changed to port 0xB779
Tftp: Sent ACK
Tftp: Response sent
Tftp: Init socket to port 0x0045
Tftp: TFTP server init done
Tftp: 	DataPort: 0xB779
Tftp: Processing packet of size 0x001F
Tftp: Set up return address
Tftp: This is block 0x0000 with opcode 0x0005 and data length 0x0013
Tftp: Error
Tftp: Response sent
Tftp: Processing packet of size 0x8613
Tftp: Overflow
Tftp: Set up return address
Tftp: This is block 0x979A with opcode 0x6D2C and data length 0xBCD4
Tftp: Invalid opcode 0x0000
Tftp: Init socket to port 0xB779
Tftp: Response sent
Tftp: Init socket to port 0x0045
Tftp: TFTP server init done
Tftp: 	DataPort: 0xB779

any ideas?

Whocareswhowins, 2016-10-25:
Closing down so no ideas

Sent from my iPhone 4s!

LoathingKernel, 2017-05-10:
This issue was moved to LoathingKernel/Ariadne-Bootloader#19

closed 2017-05-10 by LoathingKernel