codebndr/Ariadne-Bootloader #36: Please specify fuses for alternative ICSP programmer.

puhycz, 2016-11-11:
Hello,

I try program this Ariadne-Bootloader to Arduino Mega2560 by [ASIX PRESTO](http://www.asix.net/prg_presto.htm) Programmer (I have only this programmer).

When I open hex file, there are some alert about Little or Big Endian:

[image]

Please help me with settings fuses. Is somewhere specify fuses for ariadne-bootloader?

[image]

I try default fuses with SPI prog. enabled, but it is not function ...

Many thanks,

Jan

puhycz, 2016-11-16:
I finally solved it with standard avrdude.

Step by step is here (in Czech language):
https://www.puhy.cz/blog/lan-bootloader-pro-arduino-mega-2560-atmel-atmega-2560/

LoathingKernel, 2017-05-10:
This issue was moved to LoathingKernel/Ariadne-Bootloader#20

closed 2017-05-10 by LoathingKernel