codebndr/Ariadne-Bootloader #38: EEPROM.put and get does not work

elik745i, 2017-03-24:
EEPROM.put and get requests does not work with newEEPROM library, can anyone fix it? or suggest the way forward...

Whocareswhowins, 2017-03-24:
Yes you are correct it does not work.

Sent from my iPhone 4s!

elik745i, 2017-03-25:
how can it be fixed? maybe someone has more programming skills to do that? I'm currently going deep into the coding to try and fix it... I need to use that feature in my project at http://www.voltrans.az/?page_id=1969

per1234, 2017-03-25:
Just use the standard EEPROM library included with the Arduino IDE but make sure to leave alone the EEPROM addresses Ariadne uses for its configuration settings. Ariadne uses EEPROM addresses 0-26 for network settings and 27-63 for the password used in the EthernetReset library. If you're not using the password then you can use EEPROM addresses 27-63 for other purposes. The NewEEPROM library just makes sure you won't use those addresses, there's no magic to it.

elik745i, 2017-03-25:
I'm agraid it does not work like that. Ariadne library is linked to newEEPROM library and even if I'm not declaring it and as you said using standard EEPROM library it's screaming compillation errors. Try this sketch and you'll see what do I ment:
`/*
  Web Server

 A simple web server that shows the value of the analog input pins.
 using an Arduino Wiznet Ethernet shield.

 Circuit:
 * Ethernet shield attached to pins 10, 11, 12, 13
 * Analog inputs attached to pins A0 through A5 (optional)

 created 18 Dec 2009
 by David A. Mellis
 modified 9 Apr 2012
 by Tom Igoe
 modified 02 Sept 2015
 by Arturo Guadalupi

 */

#include <NetEEPROM.h>
//#include <NewEEPROM.h>
#include <EEPROM.h>
#include <SPI.h>
#include <Ethernet.h>
#include <EthernetReset.h>

EthernetReset reset(8080);
EthernetServer server(80);

// Enter a MAC address and IP address for your controller below.
// The IP address will be dependent on your local network:
byte mac[] = {
  0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED
};
IPAddress ip(192, 168, 1, 128);

// Initialize the Ethernet server library
// with the IP address and port you want to use
// (port 80 is default for HTTP):

void setup() {
  
  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // start the Ethernet connection and the server:
  Ethernet.begin(mac, ip);
  server.begin();
  Serial.print("server is at ");
  Serial.println(Ethernet.localIP());
  int k = 3000;
  EEPROM.put(100,k);
  reset.begin();
}

void loop() {
  reset.check();
  // listen for incoming clients
  EthernetClient client = server.available();
  if (client) {
    Serial.println("new client");
    // an http request ends with a blank line
    boolean currentLineIsBlank = true;
    while (client.connected()) {
      if (client.available()) {
        char c = client.read();
        Serial.write(c);
        // if you've gotten to the end of the line (received a newline
        // character) and the line is blank, the http request has ended,
        // so you can send a reply
        if (c == '\n' && currentLineIsBlank) {
          // send a standard http response header
          client.println("HTTP/1.1 200 OK");
          client.println("Content-Type: text/html");
          client.println("Connection: close");  // the connection will be closed after completion of the response
          client.println("Refresh: 5");  // refresh the page automatically every 5 sec
          client.println();
          client.println("<!DOCTYPE HTML>");
          client.println("<html>");
          // output the value of each analog input pin
          for (int analogChannel = 0; analogChannel < 6; analogChannel++) {
            int sensorReading = analogRead(analogChannel);
            client.print("analog input ");
            client.print(analogChannel);
            client.print(" is ");
            client.print(sensorReading);
            client.println("<br />");
          }
          client.println("</html>");
          break;
        }
        if (c == '\n') {
          // you're starting a new line
          currentLineIsBlank = true;
        } else if (c != '\r') {
          // you've gotten a character on the current line
          currentLineIsBlank = false;
        }
      }
    }
    // give the web browser time to receive the data
    delay(1);
    // close the connection:
    client.stop();
    Serial.println("client disconnected");
  }
}
`

per1234, 2017-03-25:
I see. You can get around that issue by using avr-libc's eeprom.h. You still won't be able to use `EEPROM.put()` but it does allow you to write or read an int with a single function call:
```arduino
#include <NetEEPROM.h>
#include <avr/eeprom.h>
#include <SPI.h>
#include <Ethernet.h>
#include <EthernetReset.h>

EthernetReset reset(8080);
EthernetServer server(80);

// Enter a MAC address and IP address for your controller below.
// The IP address will be dependent on your local network:
byte mac[] = {
  0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED
};
IPAddress ip(192, 168, 1, 128);

// Initialize the Ethernet server library
// with the IP address and port you want to use
// (port 80 is default for HTTP):

void setup() {

  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // start the Ethernet connection and the server:
  Ethernet.begin(mac, ip);
  server.begin();
  Serial.print("server is at ");
  Serial.println(Ethernet.localIP());
  int k = 3000;
  eeprom_update_word((uint16_t*)100, k);
  reset.begin();
}

void loop() {
  reset.check();
  // listen for incoming clients
  EthernetClient client = server.available();
  if (client) {
    Serial.println("new client");
    // an http request ends with a blank line
    boolean currentLineIsBlank = true;
    while (client.connected()) {
      if (client.available()) {
        char c = client.read();
        Serial.write(c);
        // if you've gotten to the end of the line (received a newline
        // character) and the line is blank, the http request has ended,
        // so you can send a reply
        if (c == '\n' && currentLineIsBlank) {
          // send a standard http response header
          client.println("HTTP/1.1 200 OK");
          client.println("Content-Type: text/html");
          client.println("Connection: close"); // the connection will be closed after completion of the response
          client.println("Refresh: 5"); // refresh the page automatically every 5 sec
          client.println();
          client.println("");
          client.println("");
          // output the value of each analog input pin
          for (int analogChannel = 0; analogChannel < 6; analogChannel++) {
            int sensorReading = analogRead(analogChannel);
            client.print("analog input ");
            client.print(analogChannel);
            client.print(" is ");
            client.print(sensorReading);
            client.println("                           ");
          }
          client.println("");
          break;
        }
        if (c == '\n') {
          // you're starting a new line
          currentLineIsBlank = true;
        } else if (c != '\r') {
          // you've gotten a character on the current line
          currentLineIsBlank = false;
        }
      }
    }
    // give the web browser time to receive the data
    delay(1);
    // close the connection:
    client.stop();
    Serial.println("client disconnected");
  }
}
```
I've never liked the NewEEPROM system. I guess it was intended to be more user friendly by hiding the issue of the EEPROM addresses used by Ariadne but I think it ends up causing more problems than just clearly documenting the address range used by Ariadne and letting the user deal with it however they like.

elik745i, 2017-03-25:
Big thanks for help, but that is just example of webserver I'm trying to get working before implementing update over the web interface into my opensourse smarthouse project. I've already built it around put and get methods and there are minimum 10 of them in the different places in the 2500 line of code (((((
and it's not just int I'm writing into the memory, but also long and char rows 
I need to be able to use get and put methods!
Is there a way to unlink that NewEEPROM library and get ariadne to work with standard EEPROM lib????

LoathingKernel, 2017-03-25:
Yes, just don't write in the first few addresses of the EEPROM, as @per1234
explained a few comments above.

per1234, 2017-03-25:
The problem is the EEPROM library's global class object name conflicts with the NewEEPROM library's global class object name, causing the error:
```
In file included from C:\Users\per\AppData\Local\Temp\arduino_modified_sketch_93640\sketch_mar25a.ino:13:0:

E:\ArduinoIDE\arduino-1.8.2\hardware\arduino\avr\libraries\EEPROM\src/EEPROM.h:145:20: error: conflicting declaration 'EEPROMClass EEPROM'

 static EEPROMClass EEPROM;

                    ^

In file included from E:\Stuff\misc\electronics\arduino\libraries\NetEEPROM/NetEEPROM.h:26:0,

                 from C:\Users\per\AppData\Local\Temp\arduino_modified_sketch_93640\sketch_mar25a.ino:11:

E:\Stuff\misc\electronics\arduino\hardware\ariadne\avr\libraries\NewEEPROM/NewEEPROM.h:47:23: note: previous declaration as 'NewEEPROMClass EEPROM'

 extern NewEEPROMClass EEPROM;
```
when you try to compile @elik745i's code. I think the easiest solution is to:
Change line 47 of NewEEPROM.h from:
```c++
extern NewEEPROMClass EEPROM;
```
to:
```c++
extern NewEEPROMClass NewEEPROM;
```
Change line 68 of NewEEPROM.cpp from:
```c++
extern NewEEPROMClass EEPROM;
```
to:
```c++
extern NewEEPROMClass NewEEPROM;
```
After doing this you would need to use `NewEEPROM.read()` and `NewEEPROM.write()` instead of `EEPROM.read()` and `EEPROM.write()` if you're using the NewEEPROM library but that shouldn't be a problem since you're not directly using that library.

elik745i, 2017-03-25:
EXCELLENT!!! thanks, it looks like it's working now! I'll try to implement changes into my project now...

elik745i, 2017-03-25:
Cool, it's working! Thanks to everyone and special to per1234

LoathingKernel, 2017-05-10:
This issue was moved to LoathingKernel/Ariadne-Bootloader#22

closed 2017-05-10 by LoathingKernel